benblazak/ergodox-firmware #59: Binary compiled on Ubuntu 14.04 does not work

hontvari, 2015-01-02:
If I compile the master branch on Ubuntu 14.04, the keyboard does nothing, the OS (Windows), does not receive any key presses. The result is the same when it is compiled on Ubuntu 12.10. 

On the other hand, the partial-rewrite branch does work as expected.

How to reproduce: 
  sudo apt-get install git make gcc-avr binutils-avr avr-libc
  git clone https://github.com/benblazak/ergodox-firmware.git
  mv ergodox-firmware/ ergodox
  cd ergodox/src
  make

Upload using Teensy Loader 1.17, reboot. The keyboard does nothing when for example J is pressed.

SvDvorak, 2015-02-15:
I have the same issue, compiling on Windows 7 SP1. After checking out some older commits I noticed that the problem disappears after checking out commit # so problem seems to be related to pull request #23.
As hontvari mentions, the partial-rewrite branch works fine.
I'm using WinAVR with Teensy Loader 1.17.

phitsc, 2015-03-05:
Same problem here when compiling under Win 8.1 or Ubuntu 14.04.

Thanks SvDvorak for that commit ID.

orcinus, 2015-05-31:
Same issue, Mac OS, checked out from master.
Additionally, all LEDs turn on and stay on.

BlinkBP, 2015-09-17:
I think that this bug is caused in \lib-other\pjrc\usb_keyboard\usb_keyboard.c