benblazak/ergodox-firmware #61: CAPS layout after layer change

kdisneur, 2015-02-18:
Hi,

I'm trying to add one layer with the most convenient programming keys (ie: `[`, `{`, `(`, `<` ...).

I was using the generator but it was not working well for the shifted keys so, on the configurator, I've defined the non shifted keys and, after, I've downloaded the source. Then I've changed the value from `kprrel` to `sshprre` when needed.

Now, each time, I hit a shifted key, all the keyboard switches in capslock and I have to hit the `shift` key to be able to hit the other keys.

example: 
* I hit `s` defined as `kprrel` => `s` # OK 
* I push the layer 1 (lpush1)  
* I hit `,` defined as `sshprre` => `>` # OK 
* I pop the layer 1 (lpop1)
* I hit `s` again => `S` # not OK

I searched for `s2kcap` but this key is not defined in my layout. Do you have some idea? I think it's stupid and I just don't understand something.

Thanks for your help

andrewwillis, 2015-02-24:
I have not had this behavior specifically, but I have had issues where I wasn't getting the key I expected. In most of my cases, it was because I didn't correctly update both the press function matrix and the release function matrix. Would you want to post your source?

benblazak, 2015-02-27:
@kdisneur: andrewwillis's suggestion sounds like a really good place to start.  I probably won't have any time to help you debug (sorry.. school and internship this semester), but i'll leave the issue open in case you'd like to post more information, and anyone else has any suggestions.

kdisneur, 2015-03-01:
Excuse me for the delay, the last week was a bit busy. Thanks for your help, you can find my mapping just below:

```C
// LAYOUT ---------------------------------------------------------------------
const uint8_t PROGMEM _kb_layout[KB_LAYERS][KB_ROWS][KB_COLUMNS] = {
// LAYER 0
KB_MATRIX_LAYER(
	// unused
	0,
	// left hand
	KEY_GraveAccent_Tilde,	KEY_1_Exclamation,	KEY_2_At,	KEY_3_Pound,	KEY_4_Dollar,	KEY_5_Percent,	KEY_6_Caret,
	KEY_Tab,	KEY_q_Q,	KEY_d_D,	KEY_r_R,	KEY_w_W,	KEY_b_B,	0x31,
	KEY_DeleteBackspace,	KEY_a_A,	KEY_s_S,	KEY_h_H,	KEY_t_T,	KEY_g_G,
	KEY_LeftShift,	KEY_z_Z,	KEY_x_X,	KEY_m_M,	KEY_c_C,	KEY_v_V,	KEY_Slash_Question,
	0,	0,	0,	0,	KEY_Period_GreaterThan,
	0,	0,
	0,	0,	0,
	KEY_LeftControl,	KEY_Spacebar,	0,
	// right hand
	0,	KEY_7_Ampersand,	KEY_8_Asterisk,	KEY_9_LeftParenthesis,	KEY_0_RightParenthesis,	0,	0,
	KEY_Dash_Underscore,	KEY_j_J,	KEY_f_F,	KEY_u_U,	KEY_p_P,	KEY_SingleQuote_DoubleQuote,	0,
	KEY_y_Y,	KEY_n_N,	KEY_e_E,	KEY_o_O,	KEY_i_I,	0,
	KEY_Backslash_Pipe,	KEY_k_K,	KEY_l_L,	0,	0x38,	KEY_1_Exclamation,	0,
	KEY_Comma_LessThan,	0,	0,	0,	0,
	0x38,	KEYPAD_Exclamation,
	0,	0,	0,
	0,	1,	KEY_ReturnEnter
),
// LAYER 1
KB_MATRIX_LAYER(
	// unused
	0,
	// left hand
	0,	0,	0,	0,	0,	0,	0,
	0,	0,	0,	KEY_LeftBracket_LeftBrace,	0,	0,	0,
	0,	0,	0,	0,	0,	0,
	0,	0,	0,	0,	0,	0,	0,
	0,	0,	0,	0,	0,
	0,	0,
	0,	0,	0,
	KEY_Spacebar,	0,	0,
	// right hand
	0,	0,	0,	0,	0,	0,	0,
	0,	0,	0,	0,	0,	0,	0,
	0,	0,	0,	0,	0,	0,
	0,	0,	0,	0,	0,	0,	0,
	0,	0,	0,	0,	0,
	0,	0,
	0,	0,	0,
	0,	0,	KEY_Spacebar
)
// ...
}

// ...

// PRESS ----------------------------------------------------------------------
const void_funptr_t PROGMEM _kb_layout_press[KB_LAYERS][KB_ROWS][KB_COLUMNS] = {
// LAYER 0
KB_MATRIX_LAYER(
	// unused
	NULL,
	// left hand
	kprrel,	kprrel,	kprrel,	kprrel,	kprrel,	kprrel,	kprrel,
	kprrel,	kprrel,	kprrel,	kprrel,	kprrel,	kprrel,	sshprre,
	kprrel,	kprrel,	kprrel,	kprrel,	kprrel,	kprrel,
	kprrel,	kprrel,	kprrel,	kprrel,	kprrel,	kprrel,	kprrel,
	NULL,	NULL,	NULL,	NULL,	kprrel,
	NULL,	NULL,
	NULL,	NULL,	NULL,
	kprrel,	kprrel,	NULL,
	// right hand
	NULL,	kprrel,	kprrel,	kprrel,	kprrel,	NULL,	NULL,
	kprrel,	kprrel,	kprrel,	kprrel,	kprrel,	kprrel,	NULL,
	kprrel,	kprrel,	kprrel,	kprrel,	kprrel,	NULL,
	kprrel,	kprrel,	kprrel,	NULL,	sshprre,	kprrel,	NULL,
	kprrel,	NULL,	NULL,	NULL,	NULL,
	sshprre,	kprrel,
	NULL,	NULL,	NULL,
	NULL,	lpush1,	kprrel
),
// LAYER 1
KB_MATRIX_LAYER(
	// unused
	NULL,
	// left hand
	NULL,	NULL,	NULL,	NULL,	NULL,	NULL,	NULL,
	NULL,	NULL,	NULL,	sshprre,	NULL,	NULL,	NULL,
	NULL,	NULL,	NULL,	NULL,	NULL,	NULL,
	NULL,	NULL,	NULL,	NULL,	NULL,	NULL,	NULL,
	NULL,	NULL,	NULL,	NULL,	NULL,
	NULL,	NULL,
	NULL,	NULL,	NULL,
	kprrel,	NULL,	NULL,
	// right hand
	NULL,	NULL,	NULL,	NULL,	NULL,	NULL,	NULL,
	NULL,	NULL,	NULL,	NULL,	NULL,	NULL,	NULL,
	NULL,	NULL,	NULL,	NULL,	NULL,	NULL,
	NULL,	NULL,	NULL,	NULL,	NULL,	NULL,	NULL,
	NULL,	NULL,	NULL,	NULL,	NULL,
	NULL,	NULL,
	NULL,	NULL,	NULL,
	NULL,	lpop1,	kprrel
)
// ...
}
```

Thanks for your help

andrewwillis, 2015-03-01:
@kdisneur could you also include the relevant snippet from the `_kb_layout_release` structure these two layers?

kdisneur, 2015-03-04:
Of course, sorry. BTW I've just seen the key for the release state is defined with `kprrel` and not `sshprre`. I'm going to try this change this evening if you don't see anything else.

```C
// RELEASE --------------------------------------------------------------------
const void_funptr_t PROGMEM _kb_layout_release[KB_LAYERS][KB_ROWS][KB_COLUMNS] = {
// LAYER 0
KB_MATRIX_LAYER(
	// unused
	NULL,
	// left hand
	kprrel,	kprrel,	kprrel,	kprrel,	kprrel,	kprrel,	kprrel,
	kprrel,	kprrel,	kprrel,	kprrel,	kprrel,	kprrel,	sshprre,
	kprrel,	kprrel,	kprrel,	kprrel,	kprrel,	kprrel,
	kprrel,	kprrel,	kprrel,	kprrel,	kprrel,	kprrel,	kprrel,
	NULL,	NULL,	NULL,	NULL,	kprrel,
	NULL,	NULL,
	NULL,	NULL,	NULL,
	kprrel,	kprrel,	NULL,
	// right hand
	NULL,	kprrel,	kprrel,	kprrel,	kprrel,	NULL,	NULL,
	kprrel,	kprrel,	kprrel,	kprrel,	kprrel,	kprrel,	NULL,
	kprrel,	kprrel,	kprrel,	kprrel,	kprrel,	NULL,
	kprrel,	kprrel,	kprrel,	NULL,	sshprre,	kprrel,	NULL,
	kprrel,	NULL,	NULL,	NULL,	NULL,
	sshprre,	kprrel,
	NULL,	NULL,	NULL,
	NULL,	NULL,	kprrel
),
// LAYER 1
KB_MATRIX_LAYER(
	// unused
	NULL,
	// left hand
	NULL,	NULL,	NULL,	NULL,	NULL,	NULL,	NULL,
	NULL,	NULL,	NULL,	kprrel,	NULL,	NULL,	NULL,
	NULL,	NULL,	NULL,	NULL,	NULL,	NULL,
	NULL,	NULL,	NULL,	NULL,	NULL,	NULL,	NULL,
	NULL,	NULL,	NULL,	NULL,	NULL,
	NULL,	NULL,
	NULL,	NULL,	NULL,
	kprrel,	NULL,	NULL,
	// right hand
	NULL,	NULL,	NULL,	NULL,	NULL,	NULL,	NULL,
	NULL,	NULL,	NULL,	NULL,	NULL,	NULL,	NULL,
	NULL,	NULL,	NULL,	NULL,	NULL,	NULL,
	NULL,	NULL,	NULL,	NULL,	NULL,	NULL,	NULL,
	NULL,	NULL,	NULL,	NULL,	NULL,
	NULL,	NULL,
	NULL,	NULL,	NULL,
	NULL,	NULL,	kprrel
),
// ...
}
```

kdisneur, 2015-03-04:
OK I changed my `_kb_layout_release` to use `sshprre` AND I also added a `lpop1` on the release button. It's now working! :heart_eyes_cat:

Thanks for your help

closed 2015-03-04 by kdisneur