benblazak/ergodox-firmware #63: cannot send some shifted keys ?

bigfish, 2015-03-12:
I'm looking at the configurator, and I don't see a way to send some shifted  keys, eg. the symbols above the numbers, or {}... I see some shifted keys, eg, +, but not all. 

Am I missing something? Is there a way to do this? I would like to have these characters triggered by single keystroke on layer. 

Sorry if I missed something obvious, I 'm just starting to look at this, while I await some supplies for finishing building my Ergodox.

benblazak, 2015-03-12:
I didn't write the configurator (though, i do think jasper, from massdrop, did a pretty good job :-) ) but i do remember some things not being available in the interface.  Notably, some of the possible keys, and also the ability (which isn't very intuitive, perhaps, but does exist) to write macros.

It's certainly possible to send any shifted key you like -- it's really just a macro that sends shift + [your key] -- but if it's not in the configurator, you'll have to do it in source.  Note that there are certain caveats with the way shifted keys work, but I don't currently know of a better way.

Bunch of people have written their own layouts in source, along with shifted keys :-), and there's been discussion on the topic in various places, so I hope you don't mind my neglecting to give further direction.  A search through the other issues here on github might at least give you someone else to ask.  Lots of stuff due at school today.  Good luck!

bigfish, 2015-03-12:
Thanks for the info... I'm glad to hear it is possible. I didn't know about the 'source' option.. I will look a bit further, and when I find a solution I will post it here to help anyone else with this question.

bigfish, 2015-04-08:
hey, so I figured out all I needed to do was switch to the partial-rewrite branch ... all the shifted symbols I want are now defined in the firmware/lib/layout/keys.h, thanks to the KEYS__SHIFTED function. I just copied and renamed your qwerty--ben.c layout, set it as default in the options.mk file, and ran 'make' from the firmware dir, and I have exactly the layout I want :) It might be nice to have a script to export a visualization of the layout based on the .c layout file, just as a convenient reference.. perhaps if I have time I will do that.

Anyhow, problem solved so I will close this issue now, and hopefully it helps someone else. Thanks for your work on this.

closed 2015-04-08 by bigfish

benblazak, 2015-04-08:
Glad you got it working :-)