benblazak/ergodox-firmware #64: Keyboard gets stuck repeating letters

pnickolay, 2015-04-13:
Hi,

I've recently started experiencing the problem where the keyboard gets stuck repeating some letter infinitely or just not accepting any input. The only fix is to replug it, but then it starts right after couple clicks again. Did anyone experienced anything like this? The leds also blink weirdly sometimes on the start (usually they just light up in order and then turn off, but now they sometimes blinks several times after that). The keyboard became unusable.

Thanks,
Nickolay

bmedicke, 2015-04-13:
Have you recently changed the firmware? With the ErgoDox it could always be hardware related. After a while I had similar issues (keys getting stuck, unresponsive rows) and had to touch up some solder joints.

pnickolay, 2015-04-14:
I did change the firmware, but last time was about couple weeks ago. Is there a way to know if it's hardware or software problem?

benblazak, 2015-04-14:
It sounds a lot like hardware.  To be sure, you might try loading a firmware that's been tested a lot (i.e. the one on dropbox, or a fresh build from either the main or partial-rewrite branches here, or one from the massdrop configurator).

pnickolay, 2015-04-14:
Thanks, will try that and tell how it goes. Btw, why does the LED blinking pattern differs on the start? Does it mean anything?

benblazak, 2015-04-14:
I forget if I put any error blink patterns in there. A quick look through `main()` should tell you :-)

pnickolay, 2015-04-14:
Hmm, interestingly the issue is gone for now just after keeping the keyboard unplugged for the night (works with same firmware). I wonder if it overheats from constant use or something similar..

benblazak, 2015-04-14:
So it's a hardware problem ;-)

pnickolay, 2015-04-16:
Shortly after that issue started to appear again. I used compressed air to clear out the keyboard and that fixed it, so i guess the issue was micro dust.

closed 2015-04-16 by pnickolay