benblazak/ergodox-firmware #66: Media Keys and F13-F19 keys ignored with Ubuntu

simao, 2015-08-10:
When using my ergodox with linux/ubuntu, my media keys and and keys from f13-19, but these keys are completely ignored by my system.

I tried getting the keycodes using `evtest` and `xev`, but these apps also cannot recognize the keys.

I use an apple keyboard and everything works there, but not with ergodox. The ergodox also works fine with Mac OS X.

Is there something I can do to make ubuntu properly recognize ergodox? Maybe some `setxkbdmap` tricks?

benblazak, 2015-08-11:
If you have keys -- especially uncommon keys like F keys greater than 12, and media keys -- that work in one OS but not in another (or, for that matter, keys that don't work anywhere) I think there's a good chance the OS keyboard driver simply doesn't implement them.  Take a look at [the USB keyboard usage page] (https://github.com/benblazak/ergodox-firmware/blob/partial-rewrite/firmware/lib/usb/usage-page/keyboard.h) if you like -- it shows which keys are supposedly (according to the USB HID spec) supported by which OSs.  To be honest, I don't really know a lot about the issue.  But the two options I can think of are:
* use normal keys (or key combinations, via macros) instead of those rare keys, and map them in the OS instead -- though this might be a pain, since you're using multiple OSs
* modify the Linux USB keyboard driver -- which would probably be best, but I'm not sure how difficult it would be

Hope you get it working!  If you have time, please post back and let me know how it went :-)

simao, 2015-08-11:
It makes sense that this is a usb driver issue. I did some more digging but couldn't find a cause for this. Ubuntu is using a generic hid driver that should implement the keycodes referenced in the document you linked.

This is my dmesg output when I connect the ergodox:

```
[243374.532733] usb 1-1: new full-speed USB device number 60 using xhci_hcd
[243374.662476] usb 1-1: New USB device found, idVendor=1d50, idProduct=6028
[243374.662479] usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[243374.662481] usb 1-1: Product: ErgoDox ergonomic keyboard
[243374.662482] usb 1-1: Manufacturer: unspecified
[243374.663483] input: unspecified ErgoDox ergonomic keyboard as /devices/pci0000:00/0000:00:14.0/usb1/1-1/1-1:1.0/0003:1D50:6028.0028/input/input112
[243374.717391] hid-generic 0003:1D50:6028.0028: input,hidraw0: USB HID v1.11 Keyboard [unspecified ErgoDox ergonomic keyboard] on usb-0000:00:14.0-1/input0
```

Is there anyone here using an ergodox with linux? What drivers are you using?

Thanks again for your help!

simao, 2015-08-14:
Hi,

I tried using the tmk firmware and the media keys work using that firmware. I don't know what it does differently but the problem seems to be in the usb drivers, which generate the codes for the tmk firmware but not for the benblazak firmware.

 I dont't know how to debug this though, I am not using any special usb drivers, I just use the kernel drivers that ship with ubuntu.

Please let me know if I can help debug this.

Thanks for your help!

benblazak, 2015-08-14:
So it really is the firmware then?  Sorry about that.  I didn't add the media keys (I tried... it was complicated... and I decided to work on other things), they were added by judascleric in pull request #23.  If you wanna take a look at his changes, and go through the relevant files in the tmk firmware to see what's different, you're welcome to :) -- and if you end up with a well tested update you'd like to send a pull request for (for the master branch) I'll gladly accept it.  Don't think I'll have time to work on that issue myself though, at least not for a long while... sorry.

muflax, 2016-01-04:
(Had the same problem, submitted a pull request that fixes it.)