benblazak/ergodox-firmware #67: can't compile project on latest version of mac osx using homebrew-avr-gcc

alexanderjeurissen, 2015-08-14:
I finished my ergodox today and altough the Massdrop keyboard configurator will suffice for creating my desired layout i do want to change some led settings and therefore want to compile the source myself.

currently I'm running the latest version of osx yosemite and installed the latest version of avr-gcc using homebrew https://github.com/osx-cross/homebrew-avr/

i downloaded the source using the massdrop keyboard configurator and tried to compile it by executing the `make` command in the directory containing the makefile.

doing so results in the following output in my terminal:

```shell
--- making main.o ---
avr-gcc -c -mmcu=atmega32u4 -DF_CPU=16000000 -DMAKEFILE_BOARD='teensy-2-0' -DMAKEFILE_KEYBOARD='ergodox' -DMAKEFILE_KEYBOARD_LAYOUT='default--layout' -DMAKEFILE_DEBOUNCE_TIME='5' -DMAKEFILE_LED_BRIGHTNESS='0.5' -std=gnu99 -Os -Wall -Wstrict-prototypes -fpack-struct -fshort-enums -ffunction-sections -fdata-sections -MMD -MP -MF main.o.dep main.c -o main.o
unknown MCU 'atmega32u4' specified
Known MCU names:
   avr2
   at90s2313
   at90s2323
   at90s2333
   at90s2343
   attiny22
   attiny26
   at90s4414
   at90s4433
   at90s4434
   at90s8515
   at90c8534
   at90s8535
   attiny13
   attiny2313
   attiny24
   attiny44
   attiny84
   attiny25
   attiny45
   attiny85
   attiny261
   attiny461
   attiny861
   at86rf401
   avr3
   atmega103
   atmega603
   at43usb320
   at43usb355
   at76c711
   avr4
   atmega8
   atmega48
   atmega88
   atmega8515
   atmega8535
   at90pwm2
   at90pwm3
   avr5
   atmega16
   atmega161
   atmega162
   atmega163
   atmega164p
   atmega165
   atmega165p
   atmega168
   atmega169
   atmega169p
   atmega32
   atmega323
   atmega324p
   atmega325
   atmega3250
   atmega329
   atmega3290
   atmega406
   atmega64
   atmega640
   atmega644
   atmega644p
   atmega645
   atmega6450
   atmega649
   atmega6490
   atmega128
   atmega1280
   atmega1281
   at90can32
   at90can64
   at90can128
   at90usb646
   at90usb647
   at90usb1286
   at90usb1287
   at94k
   avr6
   atmega2560
   atmega2561
   avr1
   at90s1200
   attiny11
   attiny12
   attiny15
   attiny28
main.c:1: error: MCU 'atmega32u4' supported for assembler only
In file included from ./keyboard/./ergodox/./controller/teensy-2-0--led.h:14,
                 from ./keyboard/./ergodox/controller.h:19,
                 from ./keyboard/controller.h:15,
                 from main.c:16:
/usr/local/bin/../lib/gcc/avr/4.1.1/../../../../avr/include/avr/io.h:332:6: warning: #warning "device type not defined"
main.c: In function 'main':
main.c:55: error: 'OCR1A' undeclared (first use in this function)
main.c:55: error: (Each undeclared identifier is reported only once
main.c:55: error: for each function it appears in.)
main.c:55: error: 'OCR1B' undeclared (first use in this function)
main.c:55: error: 'OCR1C' undeclared (first use in this function)
main.c:55: error: 'DDRB' undeclared (first use in this function)
main.c:122: warning: suggest parentheses around assignment used as truth value
main.c:125: warning: suggest parentheses around assignment used as truth value
main.c:128: warning: suggest parentheses around assignment used as truth value
make: *** [main.o] Error 1
```

i'm not really a C expert but i'm sure i can figure out the warnings. however I can't wrap my head around the error that is causing the build to fail. 

could you help me get this up and running ? or point me in the right direction ?

thanks in advance.

alexanderjeurissen, 2015-08-14:
i thought that maybe it was the homebrew version of `avr-gcc` that was causing the trouble however i uninstalled the homebrew version and installed AVR CrossPack and the problem still persists

i went ahead and dumped the version of `avr-gcc` which returned `4.1.1`  
I also tried to compile the blinky example code found on http://pjrc.com/teensy/gcc.html however i got the same error stating that the specified MCU is unknown

benblazak, 2015-08-14:
I'm using the version from macports (4.9.1) -- I think the others might indeed be too old.  I'm using the latest OS X also.

I just tried compiling some source downloaded from the Massdrop configurator.  I did have to change one of the lines in the makefile-options file (for some reason it was referencing a layout that didn't exist, so I told it to use `qwerty-kinesis-mod`), but after that it seemed to work.

[image]

alexanderjeurissen, 2015-08-14:
ok that's great I'll try that tomorrow morning and let you know how it went.

alexanderjeurissen, 2015-08-25:
It was indeed the version of avr-gcc that was out of date. This issue can be closed now.

closed 2015-08-25 by alexanderjeurissen