benblazak/ergodox-firmware #69: Will a non-soldered teensy be recognized as a keyboard?

Wesley-Chan, 2015-10-26:
Hello.

I just want to make sure my teensy is good before soldering.

I flashed the blinky firmwares into it. It flashes differently. But when I flash the ErgoDox [firmware](https://www.dropbox.com/sh/8bbol6fkvydmtmg/QLudrdEyc9) and plug it in, neither Windows Device Manager nor `lsusb` in Linux list it as a keyboard device.

I should note that my teensy is not soldered on the ErgoDox PCB yet.

Is is normal or should I be concerned? What test should I do for verification?

Thank you.

benblazak, 2015-10-26:
Take a look at the circuit diagram ([master](https://github.com/benblazak/ergodox-firmware/blob/master/src/keyboard/ergodox/circuit-diagram.svg), [partial-rewrite](https://github.com/benblazak/ergodox-firmware/blob/partial-rewrite/firmware/keyboard/ergodox/controller/circuit-diagram.svg)).  Minimally, the SCL and SDA pins (PD0 and PD1) need to have pull up resistors, otherwise, if I'm remembering correctly, the code that tries to initialize the MCP23018 (whether or not it's actually there) hangs.

Wesley-Chan, 2015-10-27:
Thanks.

I removed the whiles in `teensy-2-0.c` and it worked! So relieved.

closed 2015-10-27 by Wesley-Chan