benblazak/ergodox-firmware #76: Is 80 key version of ergodox supported in your code ?

b0f00nmanjaro, 2016-03-24:
Hi, 

I have an 80 key ergodox keyboard, does your code support 4 additional keys ?

b0f00nmanjaro, 2016-03-25:
I belive it is k15 k16 k17 k18 are the one I need to work. Hmm. I fount it in matrix.h file.

benblazak, 2016-03-26:
Good to go then?  It's been a while since I wrote it, but it sounds like you're looking in the right place :-)

b0f00nmanjaro, 2016-03-27:
I managed to change the layout the way I need it also enabled the 4 additional keys, but now 2 aditional keys are not working as it should. What happens is that when I press the key, the key behaves like it didnt get the command to stop being pressed. For example if I press the letter "R" then this letter keeps repeating it self. Do you maybe know is this a faulty switch or a programming error or wrong debounce time ? What is the recommended debounce time for greetech brown switch ? I tried 1ms and 10 ms and 5 ms, but it doesnt make a difference my key still repeats it self. Because the key repeats other keys cant be pressed. I have to unplug the keyboard and plug it in again then the reast of the keys work, but not the 2 keys.

benblazak, 2016-03-27:
Were the keys repeating before you changed the firmware?  Also, what branch are you using?  It sounds me like you're using "master" and you misplaced something in the layout matrices (which is really easy to do, which is why I changed that in "partial-rewrite").

b0f00nmanjaro, 2016-03-27:
The keys were not functional before the firmware change, it had 0 istead of a function. I am using master branch. Should I try to compile with your new code, with partial-rewrite ?

b0f00nmanjaro, 2016-03-27:
OK I tried qwerty--ben layout. the addtional keys have "nop" values. This means no option ? If I change "nop" value to "arrowL" value in qwerty--ben layout this seems it is not enough because nothing happens. What I mean is that the "arrowL" key doesnt work on the new position. In the master brench this was different, I had to change the value for the key on two positions in the layout file. Anyway my question is, is it enough to just change the nop values to some key, or do I need to do changes in some other files too ?

b0f00nmanjaro, 2016-03-27:
OK so now if I assign "F" to the key. it works and it doesnt reopeat it self. :)
Will try again the arrowD and report what happens.

b0f00nmanjaro, 2016-03-27:
Okey it works ! :)
partial-rewrite branch is working better for me, I probably made a mistake in master brach layout files, still dont know what went wrong. But whats important all 4 additional keys are working. YES ! :)
I am very happy !
Would also like to thank you for writing this code for me and other people. You are really great ! :)
Bye !

benblazak, 2016-03-27:
Glad you got it working!

closed 2016-03-27 by benblazak

perheld, 2016-10-09:
@b0f00nmanjaro : What did you end up changing to get 80 keys to work?