benblazak/ergodox-firmware #78: Latch Keys Not Working

F1LT3R, 2016-08-01:
Latch keys do not seem to work as expected.

[image]

No matter what key I set above `Left Ctrl` (`Latch 2` in this image), I get the `Latch 2` kicking in, but instead of it being latched  until the next key is pressed, it is only latched until I let go of the key. (This is how I would have expected (and want) the latch to work.)

**Desired Behavior**

```
Key down = Latch on
Key up = Latch off
```

The actual behavior when setting latch keys in other places on the keyboard is as follows:

**Actual Behavior**

```
Key down = Latch on
Key up = < nothing - happens >
Next key press = Latch off
```

I can't find any documentation on how the latch keys actually should be working, and they do not seem consistent across the keyboard.

Is this a hardware limitation or is this something that is handled at the firmware level?

benblazak, 2016-08-01:
It sounds like you're looking for a "shift" key or something similar, rather than a latch key -- but I haven't looked closely at how input club defines "latch" keys.

It does sound like a firmware / configurator issue you're dealing with though, rather than a hardware or USB HID spec issue.

Also, please keep in mind that anything written by inputclub was probably done by someone who knows more than I do, and is more active currently in keyboard related projects :-).

F1LT3R, 2016-08-01:
Ok thanks @benblazak. You are right, I am looking for a shift-like behavior. I'm currently looking into learning C and working with the [QMK Firmware](https://github.com/jackhumbert/qmk_firmware) to get the Ergodox to where I want it to go. As far as I can tell from reading docs, code and issues, QMK Firmware team might have a decent focus and pace of iteration.

benblazak, 2016-08-01:
Good luck!  Embedded stuff can be tedious, but in the end very rewarding.

And hope it works out with the QMK firmware.  I haven't kept up with all the firmwares supporting the ErgoDox, but at some point I should probably link some of the more actively developed ones here.

closed 2016-08-01 by benblazak