benblazak/ergodox-firmware #81: "Simple macros" don't work on Mac OS Sierra

rjhilgefort, 2016-10-10:
I just upgraded to OSX Sierra to find that the "simple macros" I had configured through the input club layout layout builder no longer fire consistently (fire once every 20 key presses or something). 

Anyone else having this issue? Is it a known issue- I couldn't find anything in the issues on this repo? Is this an issue I should be posting to the input club configurator instead?

rjhilgefort, 2016-10-10:
Wasn't sure where the appropriate place was, so I created an issue over there as well:

https://github.com/kiibohd/KiiConf/issues/63

benblazak, 2016-10-11:
Are you using an Infinity then?  This firmware only runs on the originals (and HaaTa, as far as I know, wrote the firmware for those) :-)

rjhilgefort, 2016-10-11:
@benblazak I am. Sorry for muddying up your issues then- I'll post something over there. Thanks!

closed 2016-10-11 by rjhilgefort

benblazak, 2016-10-12:
No worries.  On the one hand it's kinda cool that you thought it was running my firmware.  On the other hand, it's cooler than you're running HaaTa's, because he knows a lot more than I do :-)