benblazak/ergodox-firmware #82: How to release the 'RightAlt' upon leaving a layer.

zorglups, 2016-10-13:
Hello,

Using your firmware on my Ergodox for the last years, I came up with something really useful on:
- a base qwerty later
- a 'control' layer (navigation + F keys and a few other "control" stuffs (like cut/past/ctrl-z/ctrl-a/...)
- a 'keypad' layer

I only miss an important thing (under Windows) : Easy and comfortable access to the 'Alt-Tab' combo.

I wanted to implement it on my 'control' layer and added this macro to ma qwerty_zorglups.c layout:
```
// Send Alt-Tab
void keys__press__m_alttab(void) {
    usb__kb__set_key(true, KEYBOARD__LeftAlt);
    usb__kb__send_report();
    usb__kb__set_key(true, KEYBOARD__Tab);
    usb__kb__send_report();
    usb__kb__set_key(false, KEYBOARD__Tab);
    usb__kb__set_key(false, KEYBOARD__LeftAlt);
}
void R(m_alttab)(void) {}
```

I then used 'm_alttab' onto the "R" key of my control layer.

The problem is that the "alt-tab" window will only flash and we only toggle between the 2 last applications.

I then changed it to:
```
// Send Alt-Tab
void keys__press__m_alttab(void) {
    usb__kb__set_key(true, KEYBOARD__LeftAlt);
    usb__kb__send_report();
    usb__kb__set_key(true, KEYBOARD__Tab);
    usb__kb__send_report();
    usb__kb__set_key(false, KEYBOARD__Tab);
    //usb__kb__set_key(false, KEYBOARD__LeftAlt);
}
void R(m_alttab)(void) {}
```

That way, the 'LeftAlt' remain down and the Alt-Tab will stay and I can press it several time to select an application in the list.

The missing piece is : Where should I insert/call the "usb__kb__set_key(false, KEYBOARD__LeftAlt);" so that it is called when I leave my control layer?

Is that in keys.part.h ?

I changed it a bit because I do not need the LED for the layers but rather use them for power and num lock. For now, it looks like:
```
KEYS__LAYER__PUSH_POP(0, 0);
#define  keys__press__lpu0l0    P(lpupo0l0)
#define  keys__release__lpu0l0  KF(nop)
#define  keys__press__lpo0l0    R(lpupo0l0)
#define  keys__release__lpo0l0  KF(nop)

KEYS__LAYER__PUSH_POP(1,1);
// Zorglups: //void P(lpu1l1) (void) { layer_stack__push(0, 1, 1); kb__led__on(1); }
// Zorglups: //#define  keys__release__lpu1l1  KF(nop)
// Zorglups: //void P(lpo1l1) (void) { layer_stack__pop_id(1); kb__led__off(1); }
// Zorglups: //#define  keys__release__lpo1l1  KF(nop)
#define  keys__press__lpu1l1    P(lpupo1l1)
#define  keys__release__lpu1l1  KF(nop)
#define  keys__press__lpo1l1    R(lpupo1l1)
#define  keys__release__lpo1l1  KF(nop)

KEYS__LAYER__PUSH_POP(2,2);
// Zorglups: //void P(lpu2l2) (void) { layer_stack__push(0, 2, 2); kb__led__on(2); }
// Zorglups: //#define  keys__release__lpu2l2  KF(nop)
// Zorglups: //void P(lpo2l2) (void) { layer_stack__pop_id(2); kb__led__off(2); }
// Zorglups: //#define  keys__release__lpo2l2  KF(nop)
#define  keys__press__lpu2l2    P(lpupo2l2)
#define  keys__release__lpu2l2  KF(nop)
#define  keys__press__lpo2l2    R(lpupo2l2)
#define  keys__release__lpo2l2  KF(nop)
```

I'm really sorry for that badly formulated question but I'm not a C programmer and your help will be greatly appreciated.

Best regards,

Pierre

benblazak, 2016-10-16:
Your question isn't badly formatted :-)

On a stylistic note real quick: it'd be more consistent to say `P(m_alttab)`, if you're going to say `R(m_alttab)` below it -- or alternately to use `keys__press__m_alttab` and `keys__release__m_alttab`.  `R(...)` and `P(...)` are macros that expand out to the longer form during preprocessing.

To answer one question, you probably should keep all your layout specific stuff in `qwerty_zorglups.c` -- though, since it's your own fork of the repo, you're quite welcome to modify `keys.part.h` (if you're doing things the same way I did them in `qwerty--ben.c` it won't matter at all to the compiler, since `#include` is the preprocessor version of cut and paste).

Your other question is a bit harder.  It depends on what you want to happen.  It sounds like you'd like to press a single key to switch programs, and press the same key multiple times to switch to a less recent program.  What's missing is what you want to do to signal that you're done selecting a program.  You could put the release on another key if you wanted.  You could use one of the timers (I don't think I have any examples of using the functionality (I don't think I was done designing it either, but I haven't touched it in a while), but see `lib/timer.h`) and have `KEYBOARD__LeftAlt` released after a specified amount of time.  Or maybe, since you've thought about it more than I have, you have other ideas.

zorglups, 2016-10-17:
Thank you for your reply (fast, as usual).

Lazy, I don't want to use another key to signal that I'm done selecting a program. Same as with the Alt-Tab, when we release the Alt, it's done.
I would like to avoid the timer too because the reason to switch programs using Alt-Tab is to be really quick.

Would there be a way to release the KEYBOARD__LeftAlt released when I leave my layer even if it is not within the qwerty_zorglups.c?

Have a nice day.

benblazak, 2016-10-21:
You happened to catch me with a relatively straightforward question, at a time when I had time.

Yes, you can.  You can write your own layer pop function, or you can change the one in `keys.part.h`.  For example, if your control layer was layer 1, you currently have `void P(lpo1l1) (void) { layer_stack__pop_id(1); kb__led__off(1); }` changed to `#define  keys__press__lpo1l1    R(lpupo1l1)` but you could change it to `void P(lpo1l1) (void) { layer_stack__pop_id(1); usb__kb__set_key(false, KEYBOARD__LeftAlt); }` instead (untested).

Or, if there will ever be a time when you're holding alt, then pressing your layer key, then pressing your alt+tab key, you may also want a variable to store the previous state of the alt key so that you can restore that state when the layer key is released rather than always releasing the alt key.  In that case, you might make a variable above wherever you're defining the layer push and pop functions, called maybe `bool lpupo1l1__alt_pressed;`, and then inside your layer push function, add something like `lpupo1l1__alt_pressed = usb__kb__read_key(KEYBOARD__LeftAlt);`, and inside your layer pop function, instead of just releasing it, put something like `usb__kb__set_key(!lpupo1l1__alt_pressed, KEYBOARD__LeftAlt);`.

Though under certain conditions that logic might be wrong too...  It might be easier to just use left alt for this purpose, and right alt for all other purposes, so that they don't conflict.  Or there are probably other workarounds.

Didn't check all this very carefully, but I hope it's a good start :-)

zorglups, 2017-01-24:
Sorry Ben. Not had time to try it yet but I'll do and report my results.