benblazak/ergodox-firmware #91: Random sticky/unresponsive

postelrich, 2018-03-07:
A few times a day the keyboard becomes unresponsive or a key gets stuck. Unplugging and plugging back in the usb usually fixes it. Working off my configuration here: https://github.com/postelrich/ergodox-firmware/tree/master/src/keyboard/dactyl

postelrich, 2018-03-14:
Had to stop using my dactyl it got so bad, changed jobs and now using a windows pc, happens like 30 times a day.

benblazak, 2018-03-14:
didn't see a way to view a diff of all your changes at once... but it kind of sounds like a hardware issue to me.  though there is a chance that the code is getting stuck somewhere... like in the i2c stuff, or in one of your layout changes.  there's very little data that comes back to the keyboard from the OS -- i think it's just the states of the 3 LEDs -- and the USB stack should work (the person who wrote it knows much more than i do about these things lol), so i wouldn't think it was anything there.  i don't really have time to help (sorry) but you could try putting in flashing LED sequences at various points in the code, to see where it's getting stuck (unless you have any of those awesome hardware debugging tools).  or you could try adapting a different firmware, like they one used by Input Club.

postelrich, 2018-03-14:
@benblazak heres a link to better see the diff https://github.com/benblazak/ergodox-firmware/compare/benblazak:fc8099e...postelrich:master

Found this thread which suggests not getting enough power at the USB. That could very well be the issue since its now plugged into some janky usb desk hub. Used to be plugged into a https://geekhack.org/index.php?topic=81079.msg2127257#msg2127257

jave, 2018-03-14:
I used to have this problem with my ergodox a lot. Then I decided to
solder all connecting cables directly to the boards, rather than using
the connectors(that is, the usb connector and the i2c connector cable)

This greatly improved the situation for me at least.

/Joakim

postelrich, 2018-03-15:
Tried plugging directly into the thin client. No dice. 

@jave I have a dactyl which is all hand wired, so likely a short or something.

benblazak, 2018-03-16:
looking at your layout, since that's most of the diff.  do all the keys having to do with layers work correctly?  also, why is there an empty `, ,` at the end of [line 227](https://github.com/benblazak/ergodox-firmware/compare/benblazak:fc8099e...postelrich:master#diff-cb13213241e91ca52a7e9be499ea3985R227)?

that's all i can think of.  then again, it's really easy to miss something in this layout format.  then again again, if it really was a layout issue, i suspect you would have noticed a pattern by now.

postelrich, 2018-03-18:
I was only using the function layer so could remove all the others. That layer was working fine. The empty space is a typo. I did notice that on windows holding down a key doesn't repeat the letter while it would repeat on mac. Don't know if that means anything. Thanks for taking a look.

gonna try cleaning that up and merging your latest master.

benblazak, 2018-03-18:
My master branch hasn't changed for a really long time. I'll be surprised if it fixes your problem, but good luck!

pnickolay, 2019-05-29:
I'm having similar issues, and it usually seems to be triggered by the wiggle of the i2c connector cable (it also gets fixed by unpluging it and replugging again, which seems to trigger a reboot). Could it be a connection issue? Is it possible to make the firmware to be more resilient on the connection losses and restores?

benblazak, 2019-05-30:
that sounds a lot like a hardware issue

it's been a long time, so i forget exactly what the tradeoffs were.  if i'm remembering correctly, i couldn't find a way in software to detect if the connection had been lost and reset it.  i believe initialization took too long to do every cycle.  and my guess is that unplugging and replugging the trrs jack causes a short, which causes the usb hub to reset the device.

the only software solution i can think of to try is changing your firmware to initialize the i2c connection every polling cycle, and see if that brings the polling rate down too low.  otherwise, checking your solder joints and trying different cables might be the way to go :)

i suppose alternately you could map a key to re-init the i2c interface, or reboot the keyboard -- and at least that would save you disconnecting->reconnecting a physical connector.  in fact, if you're using my default qwerty layout, i believe there's already a key mapped to rebooting the keyboard (i did that a lot, for flashing new versions of the firmware) -- but again, it's been a long time.

good luck!

pnickolay, 2019-05-31:
I see, thank you! Assigning the key would not work since the keyboard is not responding at the time when it gets stuck. Will try to use a different cable fist, and then resolder the connections.