benblazak/ergodox-firmware #92: Always switched to layer 2 on the first system startup

Neur1n, 2020-02-20:
I don't know how to make a proper title, because it seems really weird to me. So I always use layer 0 with a computer of Windows 10, and I turn off it when I finish my work. The second day I start the computer for the very first time, the keyboard always switches to layer 2 (the last layer) automatically. Previously I plugged this keyboard into a linux machine, everything worked fine. I thought it might be the big-endian/little-endian issue, so I erased and download the firmware on Windows again. But nothing changed.

Any suggestion is much appreciated.

benblazak, 2020-02-20:
Don’t have time to look into it right now — but my first thought is that maybe it has something to do with the way caps lock (or another key like that) is handled?

Neur1n, 2020-03-03:
Additionally, it seems that only the right part is working during boot. I can tell it since I cannot switch the layers using the keys on the left part, and nothings happens when I try to type something with the left part.
[image]

benblazak, 2020-03-03:
Might be a hardware problem, if you haven’t double checked that yet.

closed 2020-04-09 by Neur1n