benblazak/ergodox-firmware #95: Keyboard doesnt boots after tried to flashing with another firmware

Thanatermesis, 2023-01-25:
I wanted to create this bug report especially to have some information on internet about this issue.

Today I was wondering how it work another firmware (qmk) so I tried to use it, what can happen? I will only need to reflash my older firmware on which I have a very good keymap...

Thats what I did, and it worked, I have put back this original firmware, my leds flashed as my own customization and my keys worked perfectly.... then when I rep-plugged the keyboard, nothing working, no startup leds, no keys, nothing... wtf? I was scared

So I tried to re-program it, replug it, reprogram, no way... the only way is: *by flashing the qmk firmware, and then this one again*, this works correctly... until you re-plug the keyboard 

Is this normal? I mean, the chip is not meant to be *entirely* rewrited when flashing? what could have changed that does that now ? this is a bit weird...

So now im forced to use the qmk firmware (this will require me a lot of time to convert my special keymap) or to... create a script that does a double flashing everytime my ergodox is disconnected lol

Well, any idea of why this is happening is welcome, anybody has seen this problem before? I can of course do more tests with my strange keyboard state if you want to

Notes: yes i tried to use the "reset" and the "eprom reset" keys from the qmk firmware, even if im not sure what exactly they does :thinking:

Thanatermesis, 2023-01-25:
From the teensy gui tool I can see this log information, which happens trying to trigger the "reboot" after to have flashed it, the device marks as disconnected :thinking: 

23:42:48.731 (loader): Reboot event
23:42:48.731 (loader): begin operation
23:42:48.733 (loader): sending reboot
23:42:48.733 (loader): begin wait_until_offline
23:42:48.783 (loader): action remove: /dev/hidraw5
23:42:48.784 (loader): USB remove /dev/hidraw5
23:42:48.784 (loader): HID/linux: device is disconnected(2)
23:42:48.784 (loader): offline, waited 1
23:42:48.784 (loader): end operation, total time = 0.052 seconds
23:42:48.784 (loader): set background IMG_REBOOT_OK
23:42:48.785 (loader): redraw timer set, image 14 to show for 1200 ms
23:42:49.985 (loader): redraw, image 9

benblazak, 2023-01-25:
bug confirmed on one of my boards.  this is really weird.

Thanatermesis, 2023-01-25:
it is, the only difference the keyboard is meant to have is an entirely different, rewrited contents, isnt? im not expert on it but i cannot get how this bug happens, the hardware is the same of course

benblazak, 2023-01-25:
i don't understand it yet either -- and i may not have time to figure it out, but i've been looking at it a bit today.

if you unplug the TRRS jack, and then unplug -> replug the USB, it will work, though it blinks an error (i'm still trying to remember what that means lol, it's been a while). if you then plug in the TRRS cable (quickly, maybe not too quickly) both sides will work.

not really a workaround, but interesting.

benblazak, 2023-01-26:
ahh -- maybe the error blinking was something i removed... the build i was flashing was really old.

benblazak, 2023-01-26:
not sure if i should have spent so long working on this today lol, but i got it!

two ways to fix

## first way

- if you comment out [these two lines](https://github.com/benblazak/ergodox-firmware/blob/partial-rewrite/firmware/keyboard/ergodox/controller.c#L26-L27) it should work

it's [this for loop](https://github.com/benblazak/ergodox-firmware/blob/partial-rewrite/firmware/lib/layout/eeprom-macro/atmega32u4.c#L743-L745) that's keeping the function from returning.

of course if you're using macros (i can't remember if i ever got them to work) this isn't the solution you want lol.

## second way

- install teensyduino https://www.pjrc.com/teensy/teensyduino.html
- run the sketch below (modified from [here](https://docs.arduino.cc/learn/programming/eeprom-guide)) to set every byte in the eeprom to 0xff

```c++
#include <EEPROM.h>

void setup() {
  // initialize the LED pin as an output.
  pinMode(11, OUTPUT);

  for (int i = 0 ; i < EEPROM.length() ; i++) {
    EEPROM.write(i, 0xff);
  }

  // turn the LED on when we're done
  digitalWrite(11, HIGH);
}

void loop() {
  /** Empty loop. **/
}
```

for whatever reason, initializing all eeprom bytes to 0x00 doesn't seem to work. definitely a bug, but not one i'm going to try and fix right now (lots to do).

thanks for catching this.