benblazak/ergodox-firmware #96: Using ancient frozen customized fork of your firmware (didn't set it up, kinda inherited it, don't really understand it). If I get a new ergodox, could I just flash my current firmware on to it? (Paranoid the old microcontroller in my current one *might* be starting to fail...)

dwawlyn, 2023-07-20:
So I've got an old ergodox from https://falba.tech/
and I'm getting kinda paranoid that the old microcontroller in it *might* be starting to fail.
(
according to the output of `lsusb`, it's a:
#=>
`ID 1d50:6028 OpenMoko, Inc. Teensy 2.0 Development Board [ErgoDox Keyboard]`
)

I have a *very* custom layout, and I'm *heavily* dependent on it,
so I'm trying to prepare an emergency replacement plan ahead of time
(while I can still type normally).

The apparently quickest and easiest thing to do would be to just get an https://ergodox-ez.com/

But my question here is like...

Well, I'm using an ancient frozen customized fork of your firmware.
(I just kinda... inherited it from... my dead partner...)

I didn't set it up
(I wouldn't've been *able* to set it up)
and I don't really understand it all.
(
There's like, a ruby script added in that actually generates the layout from an custom DSL
which is basically an ASCII-art rendition of the layout in a heredoc...
... and I don't even *know* what other little tweaks there might be.
)

So my question here is:

If I got a new ergodox-ez,
would I be able to just flash the firmware I'm currently using with my old ergodox onto the new?

Like, just flash it with the `firmware.hex` file?

Or would I have to load something else onto the new ergodox first,
and *then* I'd be able to do that?
(
And if I *would* have to load "something else" onto the new ergodox first
(before I could flash it with my current `firmware.hex` file),
would that "something else" just be something I already *have* in the directory I have here
(for this ancient frozen customized fork of the firmware)?
)

Thank you...

benblazak, 2023-07-20:
Best to check with them, but I believe the ErgoDox EZ (at least the original, with no extra LEDs) is firmware compatible with the original ErgoDox boards. Not familiar with falba.tech, or any changes they might have made. If I recall, `lsusb` output shows a string from the firmware, rather than anything about the microcontroler (or of course the way the switches and such are wired) which is what determines firmware compatibility, *especially* if you can't change settings (e.g. for diode direction) and recompile.

I'd be surprised if the controller is going bad, but less so if some of the solder joints or switches were. In any case a backup plan is always a good idea :) -- (sorry to hear about your partner).

If indeed the boards are firmware compatible, the I believe flashing the `.hex` is all you'll have to do.

Have you looked at the ErgoDox EZ online configurator? If it's possible to duplicate your layout there, and you don't want to fuss with code, I recommend it. I recently moved to using it myself actually, since my switches are getting old, and I didn't want to go back and implement a proper debounce in my firmware right now.

dwawlyn, 2023-07-23:
Thank you!

Yeah, falbatech really aren't relevant to me anymore
-- I think they're probably *great* to have for people in Europe
(especially now they seem to be bigger and more settled in, so I imagine they've probably gotten their quality-control way up since the earlier days I dealt with them)
but I'm not in Europe anymore, so yeah.

But I *could* easily email them to ask if they can tell me about any relevant details that might differ between my keyboard and an ergodox-EZ,
if I knew what specific questions would actually *be* the relevant details to ask about...?

Like, I guess I should email them something like this?:

> Hi, I got an ergodox from you back at the end of 2015 (around October I think).

> I'm not in Europe anymore,
> and to prepare for any emergency failure of my old falbatech ergodox,
> I need to figure out how to transfer my firmware/layout cleanly between it and an ergodox-EZ or something.

> So could you tell me any relevant details you think you can remember about the details of my 2015 falbatech ergodox?

> Like, anything about the microcontroller,
> the way switches are wired etc,
> diode directions...?
> Any way in which it might differ from the standard starting-point ergodox design...?
> (Any diagrams?)

> I don't really understand the kind of technical hardware details I'm asking about here myself,
> but I figure I should be able to use them to confirm with someone who does.

Anything I should add/change/clarify?

---
Other than `lsusb`, I only know about `lshw` and `xinput`. Do you happen to know of any other maybe-relevant tools I should keep in mind?

---
> In any case a backup plan is always a good idea :)

Yeah, I know, and I should *really* have gotten started on it like a year or two ago by now,
but... well, it was really difficult to focus on when I kept feeling like...
there's one really simple solution that would mean I'd *never* have to worry about backup plans again,
and *maybe* the only reason I have left *not* to use that instead is just... irrational animal fear...
So, it was easier to keep myself distracted with less difficult things instead...

---
Yeah, I did already check with the ergodox-EZ people,
but they only have a customer-service type of email,
and the person I got was understandably not really equipped to deal with such a weird, practically unique complicated technical question...

There *was* that subreddit
https://reddit.com/r/ergodox
which functioned as their defacto community
(I think there were a lot of people there who weren't even aware that the ergodox design itself and the ergodox-EZ/ZSA-the-company were different things!)
but with the reddit API-calypse
[(*... heh, I just came up with that literally two seconds ago and I'm stupidly lowkey proud of that pun now... *someone* must've independently invented it already...*)]
I'm not sure what there is left...

I asked the customer-service person if they had any other community places left,
and they just said basically like
	- "OLKB subreddit?"
	- "QMK Discord?"
and both of those seemed less like the best-next-thing-to-try than asking here.

---
---
> Have you looked at the ErgoDox EZ online configurator? If it's possible to duplicate your layout there, and you don't want to fuss with code, I recommend it.

Yeah, that "[oryx](https://configure.zsa.io/)" thing?
(They also recommended I give a look at that.)

I *did* try using it ages ago,
[... 72 lines cut ...]
and I *think* it was shorting out
(I remember one of the diodes on the microcontroller would actually softly *glow* when I touched a particular pair of contacts while trying to clean it??).
Then it finally failed completely,
and, having nothing else to lose,
I just *doused* the microcontroller in rubbing alcohol and cleaned it that way.
-- Miraculously, it started working again, and I didn't have any problems with it after that.

It's only now, after my other old laptop unrelatedly exploded a few days ago,
and I had to plug it in to this old laptop instead,
that I started getting problems again with the ergodox not automatically coming back when I wake the laptop from sleep/suspend/whatever-it's-called.

However, I'm suspecting that it *might* be a purely software issue with this system
(it *is* plugged in through an old usb hub/splitter/dongle thing...)
and this system (opensuse tumbleweed) does have other driver issues with the interactions between old hardware
(
	eg I just could *not* get the old integrated radeon graphics and VGA external monitor support to work together,
	and I know *that's* a software thing cuz the VGA external monitor worked fine in a MX-linux live usb on the same laptop
),
sooo... yeah.

Like I said, I don't want to risk plugging and unplugging it a bunch just to find out...
(
And I haven't even *wanted* to sleep the laptop in the past few days anyway,
since I've been running so many big backup/copy jobs with my old slow external harddrives...
)

benblazak, 2023-07-24:
Hmm -- I would ask about diode direction, and any changes vs the original ErgoDox (which is the same as the original ErgoDox EZ).

For figuring stuff out yourself, rather than any command line tools, you'll likely have to open up the board (just the case). Might be worth taking pictures, so you don't have to open it again later. Might also be worth figuring out how to compile the firmware yourself, in case you need to -- which isn't the easiest thing if you haven't done it before, but with some time and patience it should be possible.

Might wanna check out those subreddits / forums though :) as I'm starting a new job tomorrow and won't have time to help after that.

Yeah oryx sounds right. I understand about missing certain functions... especially double shift == caps lock. You can kind of work around some of them -- I currently have caps lock on the same key as shift, but on a different layer ([layout here](https://configure.zsa.io/ergodox-ez/layouts/3ovvz)) -- which isn't as good, but not worth building a custom firmware to work around, for me. I believe the F13... keys are available now -- they seem to still be adding features from time to time. And of course layers are well supported. Don't remember about assigning multiple key codes to a single keys.

It is a little confusing asking for help with it, because it uses the QMK firmware under the hood, which runs on many boards and is quite capable, and you could in theory download the source, modify it, and compile it yourself -- many people do. If you do end up trying that, they have lots of documentation. I haven't taken the time to try it though, as the layout i linked above works well for me (and i don't want to maintain my own little fork right now).

Ahh, yes, sleep/wake, USB in general, and caps lock indicators, among other things, are magical, and I'm not sure any of them will work right on any given system. The QMK firmware is more likely to work right in these types of situations, as it's more widely used, and more actively maintained.

On that note, depending on how much time vs money you have, you probably could find someone to help, either with the physical board or the firmware, or both. I recall seeing lots of skilled electronics people on e.g. https://www.reddit.com/r/MechanicalKeyboards/ (or if not there, they could probably point you in the right direction).

Good luck!

dwawlyn, 2023-07-24:
> Might wanna check out those subreddits / forums though

Oh yeah, that's definitely where I'm going *next*. I just figured I should check here *first* if there was anything specific to what I'm starting with.

> Might also be worth figuring out how to compile the firmware yourself, in case you need to
> -- which isn't the easiest thing if you haven't done it before, but with some time and patience it should be possible.

By "compile the firmware", do you just mean running `make` in the dir with the `src` subdir?
(Since that's already what I do when I want to update the `firmware.hex` file after modifying the layout.)
Or something more complex?

Actually,
one thing I'm still not sure about is:
is there only a single "layer" to firmware?
(ie, you flash the hex file onto the microcontroller and that's it.)

Or is there some sort of "double-layer" thing,
where there was something *first* loaded onto the microcontroller,
which controls/affects how it actually deals with the hex file when you load it on there?

benblazak, 2023-07-24:
:)

Oh, nope, that's it, you're already recompiling the firmware. In that case, you could also change various options (depending on what version of the code you're using) if you need to e.g. reverse the expected diode direction, before compiling. So it's just bigger changes (different microprocessor, different IO expander, different switch wiring layout; which are less likely to have been done) that you have to worry about.

There is indeed a bootloader, preloaded, that interfaces with the teensy loader to load the firmware (the .hex file) -- https://www.pjrc.com/teensy/first_use.html

dwawlyn, 2023-07-24:
Ah. But that bootloader isn't one of the things that gets generated in the `make`.

I mean, I do remember flashing a hex for a layout I generated with oryx onto the same keyboard like a couple years ago,
and it worked fine (at least within the limitations I explored...),
so... I'm just working on the level of those hexfiles,
not tweaking anything more fundamental than that, right?

I'm just... trying to figure out what I should ask now so I can figure out the rest myself later
(but I'm feeling particularly tired and dumb right now).

Like, *what* parts to look up *where*...

I've got the old copy of this git repo,
with the code (and a bit of documentation, ie a few README's)
and that ruby script I mentioned,
which I know *must* generate the bits of C code that actually set the layout,
so I should be able to figure out a lot from that...

Other than that there's the documentation for the teensy itself...

... and I guess *maybe* the documentation for some of the dependencies I needed to build it?
(just like `avr-libc` and `cross-avr-gcc12` I think.)
((Well, also `libpng12-0` and `libusb-0_1-4`, but those were just dependencies for running the teensy flashing program I think.))

But I *think* I'm just working on the level of the code I have in the old copy of this repo, right?

... okay yeah, I am feeling *way* too tired and dumb right now.
I'll have to get back to this later, figuring out... what I even need to figure out.

benblazak, 2023-07-24:
> Ah. But that bootloader isn't one of the things that gets generated in the make.

correct

> I mean, I do remember flashing a hex for a layout I generated with oryx onto the same keyboard like a couple years ago,

then your board is probably firmware compatible with the ergodox ez

> I've got the old copy of this git repo,

yeah, i think you're wayyyy further along than you think. it sounds like you actually have everything you need.

dwawlyn, 2023-07-26:
I did just email falbatech,
and man their selection has grown and gotten fancier since back when I got mine.
https://falba.tech/wired-keyboards/

Makes me wish I *was* still in Europe...

They even have *wireless* ergodoxes
(well, "Redox" and "ErgoDash", but yeah.)
https://falba.tech/wireless-keyboards/

I didn't know you could *buy* anything like a wireless ergodox *anywhere*?

dwawlyn, 2023-07-27:
Oh yeah, I'm definitely not gonna be trying to *get* one myself, but just like, huh, everything sure seems a lot less limited than back when I first got my first ergodox. I guess I'm still miscalibrated on how hard keyboard stuff is to deal with in general from all the trouble/limitations I ran into back then...