benblazak/ergodox-firmware #97: default Make target (dist) fails running gen-ui-info.py with KeyError: '_kb_layout'

Apteryks, 2023-09-03:
Hi,

Running `make` using the latest commit gives me:

```
$ LANG=C make
git commit -a
On branch master
Your branch is up to date with 'origin/master'.

Untracked files:
  (use "git add <file>..." to include in what will be committed)
	build/

nothing added to commit but untracked files present (use "git add" to track)
make: [makefile:65: checkin] Error 1 (ignored)
rm -r 'build/ergodox-firmware--master--20181228T191847--89b7e2b--qwerty-kinesis-mod'*
mkdir -p 'build/ergodox-firmware--master--20181228T191847--89b7e2b--qwerty-kinesis-mod'
cd src; make LAYOUT=qwerty-kinesis-mod all
make[1]: Entering directory '/home/maxim/src/ergodox-firmware/src'

---------------------------------------------------------------
------- done --------------------------------------------------

avr-size --target=ihex firmware.hex
   text	   data	    bss	    dec	    hex	filename
      0	   9454	      0	   9454	   24ee	firmware.hex

avr-size --target=ihex firmware.eep
   text	   data	    bss	    dec	    hex	filename
      0	      0	      0	      0	      0	firmware.eep

you can load "firmware.hex" and "firmware.eep" onto the
Teensy using the Teensy loader

---------------------------------------------------------------

make[1]: Leaving directory '/home/maxim/src/ergodox-firmware/src'
cp 'src/firmware.hex' 'build/ergodox-firmware--master--20181228T191847--89b7e2b--qwerty-kinesis-mod/firmware.hex'
cp 'src/firmware.eep' 'build/ergodox-firmware--master--20181228T191847--89b7e2b--qwerty-kinesis-mod/firmware.eep'
cp 'src/firmware.map' 'build/ergodox-firmware--master--20181228T191847--89b7e2b--qwerty-kinesis-mod/firmware.map'
( ./'build-scripts/gen-ui-info.py' \
	--current-date '2023-09-02 21:52:00-04:00' \
	--git-commit-date '2018-12-28 16:18:47 -0800' \
	--git-commit-id '89b7e2bfdafb2a87e0248846d5c95cc5e9a27858' \
	--map-file-path 'build/ergodox-firmware--master--20181228T191847--89b7e2b--qwerty-kinesis-mod/firmware.map' \
	--source-code-path 'src' \
	--matrix-file-path 'src/keyboard/ergodox/matrix.h' \
	--layout-file-path \
		'src/keyboard/ergodox/layout/qwerty-kinesis-mod.c' \
) > 'build/ergodox-firmware--master--20181228T191847--89b7e2b--qwerty-kinesis-mod/firmware--ui-info.json'
Traceback (most recent call last):
  File "/home/maxim/src/ergodox-firmware/./build-scripts/gen-ui-info.py", line 476, in <module>
    main()
  File "/home/maxim/src/ergodox-firmware/./build-scripts/gen-ui-info.py", line 467, in main
    dict_merge(output, gen_mappings( args.matrix_file_path,
  File "/home/maxim/src/ergodox-firmware/./build-scripts/gen-ui-info.py", line 397, in gen_mappings
    parse_layout_file(layout_file_path) )
  File "/home/maxim/src/ergodox-firmware/./build-scripts/gen-ui-info.py", line 376, in parse_layout_file
    [[eval(el) for el in layer] for layer in layout['_kb_layout']]
KeyError: '_kb_layout'
make: *** [makefile:79: build/ergodox-firmware--master--20181228T191847--89b7e2b--qwerty-kinesis-mod/firmware--ui-info.json] Error 1
```

Is this a bug, or ?

Thanks :-)

benblazak, 2023-09-03:
works on my machine :) (lol) with a fresh clone -- can't tell what might be wrong here

Apteryks, 2023-09-03:
Which version of Python do you use?  Here it's at 3.10.7.

benblazak, 2023-09-03:
i'm on 3.11.3 right now, but the last change was ~10 years ago.

Apteryks, 2023-09-19:
another small thing: 

```
$ make clean
git clean -dX  # remove ignored files and directories
fatal: clean.requireForce defaults to true and neither -i, -n, nor -f given; refusing to clean
```